spoter368/os3m-hardware #1: Document software needed and BOM

MarcusWolschon, 2023-11-21:
Please document what software is needed to open the files.
My Alibre Design/3D Systems  Geomagic Design uses SLD_ASM and SLD_PRT too but refuses to open the files.

A Bill Of Meterials especially for the foreign coins needed would be useful too.

spoter368, 2023-11-21:
Just added STEP and 3MF files as well! SLDPRT and SLDASM files are for Solidworks, my CAD program of choice (though I know it's not the most open source friendly)

Kadah, 2023-12-11:
A BOM for the PCB components would be cool. I see #6 was just added submitted today, how convenient. :+1:

MarcusWolschon, 2023-12-13:
> Just added STEP and 3MF files as well! SLDPRT and SLDASM files are for Solidworks, my CAD program of choice (though I know it's not the most open source friendly)

Thanks.
However, I suggest to add that info to the .md instead. ;)

spoter368, 2023-12-27:
No, unfortunately the type-C connector I used has no equivalent on LCSC, so a redesign is needed for JLC to be able to build it. I plan on doing another hardware redesign as soon as this weekend to rectify that issue, though, so keep your eyes open for updates.

moth4017, 2023-12-27:
> No, unfortunately the type-C connector I used has no equivalent on LCSC, so a redesign is needed for JLC to be able to build it. I plan on doing another hardware redesign as soon as this weekend to rectify that issue, though, so keep your eyes open for updates.

thanks  for the update , there are a few of us who  have 3d printers  would like to build them , so  as soon as available from JLCPCB we are keen in ordering,  thanks for the reply .

moth4017, 2024-01-02:
do your updates get posted on your logs?

spoter368, 2024-01-02:
No, I just have been responding to issues here as needed. I updated the PCB this weekend so all parts can come from JLC only to find out LCSC is out of the LDC1612! I want to validate the PCB before I commit it here though so people don’t build defective systems. Will keep this ticket updated

JX5S, 2024-01-02:
LDC1612 is in stock, C202191, 289pcs. It's the LDC1614 that's a bit more problematic, only 67pcs

spoter368, 2024-01-03:
Ah, thank you! I promise they were out of stock this past weekend. I just placed my order for the updated board, and if it comes back and works, I will update the repo

spoter368, 2024-01-13:
@moth4017, I just published a release with the files needed to be assembled at JLCPCB

moth4017, 2024-01-15:
cheers Mark

Llybel, 2024-01-17:
> @moth4017, I just published a release with the files needed to be assembled at JLCPCB

Does this include the larger MCU your logs meantion needed? If not the stm32f103 is 64kb and is a preferred part for JLC which means it doesn't get the extended part fee added to PCB assembly costs.

spoter368, 2024-01-17:
Yes, this includes the larger MCU my logs mention. The STM32F103 is not an ideal choice for this project for a number of reasons, one being that it needs an external crystal to use the USB peripheral, rather than the STM32F042 I chose, which can use its internal clock for USB. Thanks for the suggestion though!