spoter368/os3m-hardware #3: Alternate inductor plane material - not US legal tender

SketchyFletchy, 2023-12-10:
Great concept! Small limitation for non-US citizens is that we don’t have convenient almost-flat discs of Nickel clad copper hanging around.

I’ll try a build just using some cutout thin copper sheet, which should do the job. I’ll give some thin iron strip a try too. Wouldn’t bother with aluminium though.

spoter368, 2023-12-10:
Thanks! I had no idea this project would be anywhere near as popular as it is, so hopefully you can forgive the use of "proprietary" materials ;)

Thin copper sheet should probably work, but will depend on the thickness. You'll need to make sure it's at least a few skin depths thick in order for the sensor to properly register its presence. My go-to resource for targets is TI app note SNOA957B, available here: https://www.ti.com/lit/an/snoa957b/snoa957b.pdf?ts=1702241246361 

Section 4.2 actually mentions aluminum as an excellent target material, so I wouldn't rule that out. Best of luck on your build!

Kadah, 2023-12-11:
I wonder if a copper washer could work. An M8 one looks to be about 2mm smaller diameter and slightly thicker than a dime.

lpurdy01, 2024-01-02:
I wanted to do some research into what the best targets would be (even in the US I found it difficult to find dimes). So I built a rig to measure distance vs LDC response and then ran a bunch of targets through test sequences. 

https://github.com/lpurdy01/os3m_target_testing

@spoter368 Based on the data I collected I made a hypothesis as to what I think the best target shape would be. The minimum order quantity from SendCutSend was 30. Email me at lpurdy01@<the google email service (gmail)>.com with an address if you want me to send you some.

During development I found non linearities that may make it very difficult to eliminate axis crosstalk in the steward platform solve. It may be a good idea to make a version with 4 sensor groups instead of 3. This would greatly simplify localization.

lpurdy01, 2024-01-29:
Some more progress on this issue! 

I have made and tested a custom target. It is a great shape. 
https://youtu.be/jGxCzWzAt6w
https://github.com/lpurdy01/os3m_target_testing

I am excited to try a knob with this target design.

Tropaion, 2024-02-25:
@spoter368 @lpurdy01 I was thinking of getting the knob slm printed in aluminium but then it occured to, the induction would probably not work anymore. What do you think? Would it be possible to get it to work?

lpurdy01, 2024-02-27:
I think it would work, but I think we have some refinement on the target distance before you would want to make something so expensive. 

The aluminum would interfere a bit, mostly in rotation. As long as you made the bottom edge of the knob (facing the PCB) relatively thin ~1mm it shouldn't have enough surface area facing the coils to matter.