spoter368/os3m-hardware #8: Printing process documentation

Zufallsgenerat0r, 2023-12-16:
Could you provide some information on material's and settings used for printing?
Great project, thank you for the work.

Tropaion, 2024-03-02:
I also would like to know that. My spring is way to flexible, my knob is wobling even if not touched.

spoter368, 2024-03-02:
Used default 0.2mm layer, speed profile for the Prusa mini PLA. Turn off Arachne perimeter generation.

hotsgit, 2024-04-03:
This flexure modded by lpurdy01 is apparently easier to print: https://github.com/spoter368/os3m-hardware/pull/5

I used PLA+ (sometimes called PLA Pro or PLA Tough) with 0.4 or 0.6mm nozzle, 0.2mm layer. The higher print temperature than PLA mean it's less likely to creep/sag at room temperature. Creep _can_ happen to PLA under continuous force (but nobody has mentioned yet that it's an issue for this mouse).

I think PLA+ is less rigid than PLA but the flexure is plenty stiff for me, and I can't see the mouse wobble when I shake my table.

PETG is too flexible/inaccurate unless someone changes the model.

Tropaion, 2024-04-04:
I tried different materials and finally printed it with PLA-CF, got the overall best results with it.

hotsgit, 2024-04-08:
I know that's good for stiffness, but have you had any creep (sagging over time just from the weight of the knob) or cracking with PLA-CF? I'd kinda expect it. Now I've found even PLA+ creeps noticeable in the sensor readings for me when I leave the mouse sitting plugged in.
[image]

so I'll either design a flexure to be made out of PETG or go a different route and use nitrile O-rings. Currently testing whether O-rings creep under various stress levels.

hotsgit, 2024-04-14:
Update, nitrile O-rings won't work as precision springs. Even at just 132% stretch for 5 days I noticed permanent deformation where it wouldn't return to the original unstretched diameter (~3% increase).

Was curious whether time affected it, so I hooked an o-ring to a spring to test creep under nearly constant force:
[image]

Guess I should test PETG springs for that too.