spoter368/os3m-hardware #9: Floating BOOT0 Pin Causing Inconsistent Boot Entry Point

lpurdy01, 2023-12-20:
I was having problems the board inconsistently entering the bootloader or the flashed firmware. 

The fix is to add a jumper that holds `PB8` (`BOOT0`) to GND. Optionally, on the next board revision a button and two resistors could be added to keep `PB8` low unless the button was pressed. This would allow the user to upload firmware to the `STM32 Bootloader` by holding the reset and `boot` buttons. It could be useful for the project to be able to distribute `.hex` files for users that don't have STLink devices to update their boards. 

[image]

spoter368, 2023-12-20:
Ah, interesting. I ran into this bug too but it locked me in the bootloader only. Instead of modifying the hardware, I flashed the option bytes to ignore the BOOT0 pin value. If you're using STM32CubeProgrammer, simply change the BOOT_SEL option byte to be unchecked. No hardware mod necessary.

[image]

lpurdy01, 2023-12-20:
Cool fix. I will be applying that to my other board, so I don't have to mod it. 

For me it was mostly only going into bootloader. I could only get it to boot normally when I fuzzed the power or held the pin low by measuring it with a multimeter. Thus the mod

spoter368, 2024-01-03:
As a note, the new revision of the PCB I just ordered (on 2024/1/2) has BOOT0 tied to GND. Once I validate this works, I'll close this issue

spoter368, 2024-01-13:
Fixed on the latest release and validated

closed 2024-01-13 by spoter368