spoter368/os3m-hardware #10: Overhauled boards

JX5S, 2024-01-02:
<p float="left">
  [image]
  [image] 
</p>

I've made several changes to the design, from what I remember:

- All components from LCSC (should be assembleable by JLC)
- Used components I prefer (button, USB-C)
- 0603 passives
- Reduced amount of component values
- Reduced amount of decoupling caps
- Changed debug header to only have power and SWD, same pinout as generic STM32 boards
- UART is on testpoints
- Set components that shouldn't be in CPL/BOM to not be there
- Changed routing
- Changed board thickness to 1.6mm and ensured there's 3D models for everything for CAD reasons
- Significantly reduced amount of DRC errors
- Added silkscreen around coils
- Added KiCad logo, changed OSHW logo

Impemented suggestions I've seen here:
- testpoints for some GPIOs
- fixed BOOT0

As a base I used master from from around 30.12.2023. Since the majority of changes are purely just my personal preference, the routing looks pretty bad, I haven't tested it and it might not be 100% compatible (it should be compatible, coils and board shape are identical, but programming header and USB-C etc aren't), should I do a pull request or just upload it to my account as an alternative board? I don't know when I'll get around to ordering it, it could be a matter of months

spoter368, 2024-01-03:
Wow! This is amazing! So glad to see others building on my work. If you validate this board, I'd be happy to call this V1.1 or something and approve a PR to make this the main board others work off of.

willie68, 2024-01-12:
That's nice. 
As already entioned in #11., i have problems with my manufacturer with the original board. Is it possible that you can send/upload the kicad file to check with aisler? ([aisler.net](https://aisler.net)

JX5S, 2024-01-15:
Sorry, got busy with other stuff. I've uploaded the board files to my github if you want to check how I got rid of most of the warnings. This isn't an issue, I'll close it

closed 2024-01-15 by JX5S