spoter368/os3m-hardware #11: Aisler PCB manufacturer: shortend pads

willie68, 2024-01-11:
Hi,
i just wanted to order the pcb via aisler.com. (My favorite PCB manufacturer) 
I uploaded the kicad PCB file and aisler checked it and found 6 shorted Pads.
Are these pads consciously connected? So is it save to order the board?
Here ist the original message:

```
We detected deviation(s) between the netlist and the processed layout of your pcb. At least one of the listed nets shares at least one pad, which is conductively connected the other net. Please check your layout before ordering. See also our [help article](https://community.aisler.net/t/smart-tests/52).
NET-(C8-PAD1), NET-(C8-PAD2)
NET-(C19-PAD1), NET-(C19-PAD2)
NET-(C20-PAD1), NET-(C20-PAD2)
NET-(C7-PAD1), NET-(C7-PAD2)
NET-(C9-PAD1), NET-(C9-PAD2)
NET-(C10-PAD1), NET-(C10-PAD2)
```

spoter368, 2024-01-12:
Hi Willie, I'd hold off on ordering the PCB just yet. I'm just finishing up debugging the latest version that can be fully assembled by JLCPCB and fixes some of the earlier problems with the board. Should be released tomorrow.

To answer your question though, yes, those pads are deliberately connected through the sensing coils on the PCB. I just am not experienced enough with KiCad to know how to tell it I did this on purpose :)

willie68, 2024-01-12:
Thanks for the reply. 
I dont use Kicad, too. I've been using Target for what feels like forever. 
But in target there is no Kicad importer, only one for eagle. 
Do you plan to use github releases for the newest version? 
(Which makes it much easier to get the right files)

spoter368, 2024-01-12:
I've never used them before, but I will do my best to use github releases going forward!

spoter368, 2024-01-13:
New PCB released via github releases workflow! Closing out this issue.

closed 2024-01-13 by spoter368